epsilonrt/ad7124 #1: Tests failed with -3

immonenantti, 2018-07-12:
Hey,
I was wondering if you could share your hardware configuration for the AD7124, particularly the input and references. I have been stuck with the ADC for some time now, even with the provided Arduino code.

Writing of registers possibly works, and I can e.g. enable external clock pin. But all ADC readings result in incorrect values with frequent read timeouts.

Thanks for your time.
Antti

epsilonrt, 2018-07-12:
Hello,
It must be understood that the inputs are differential.
Typically a timeout problem is caused by miscommunication on the SPI
bus. Most of the time, the problem is hardware (signal SS does not
arrive on the pin of the circuit AD7124 for example).
[ad7124-arduino-shield.DSN.pdf](https://github.com/epsilonrt/ad7124/files/2187342/ad7124-arduino-shield.DSN.pdf)

You will find attached the diagram of the shield that I designed for my
students.
Have a nice day

immonenantti, 2018-07-13:
Thank you very much for your help, solved the problem.

closed 2018-07-13 by immonenantti

epsilonrt, 2018-07-14:
Hi,
It would have been nice to take a moment to explain the cause of the problem in order to benefit everyone.
Thank you.