epsilonrt/ad7124 #15: ESP32 with EVAL-AD7124-8-pmdz

ziatajo, 2021-03-13:
Hi,
I'm trying to use an Adafruit Feather ESP32 with the EVAL-AD7124-8-pmdz ( [https://www.analog.com/en/design-center/evaluation-hardware-and-software/evaluation-boards-kits/eval-ad7124-8-pmdz.html#eb-overview]).

I used the 8-multi-continous example and reduced it to just one channel. This worked so far, I can read the data. 
I configured the Sinc4Filter for the channel and tried to increase the Sampling rate.
I tested different Filterwords:
FW=240 -> 80Hz
FW=120-> 160Hz
FW=60 -> 180Hz

So for Filterwords above 120 I get the Sampling rate according to the datasheet.
But as soon as I get lower I don't get the expected Sampling rate (I think for FW=1 it was ~200Hz)

What could cause this problem? I think somehow the SPI isn't working as it should.

`/* Initialize the SPI communication. */
  ret = drv.init (slave_select, false, 1000000, 1, 0);`

I tried it with a Teensy 3.2 first(sadly I didn't test small FW) and this code in private.cpp worked.
To make it work with the ESP32 I had to change it to 4000000.
I actually don't know what I'm doing here, but somehow it helped to get some data...
My guess was that the SPI frequency is too low, to transfer the bigger amount of data. Could that be? 
When I use my multimeter to measure the Frequency of the SCLK pin. I get only ~68kHz... Shouldn't this be the 4MHz? If I change the 40000000 to other working numbers (just try and error...) the frequency on the pin changes but not that I could detect a pattern... and it never goes into the Mhz range...

If you have some ideas what could be causing the problem, I would very much appreciate it.

Best regards.

evermore99, 2021-03-13:
Hi ziato,

I had a long conversation also with analog devices. You could use two seperate SPIs, instead running over the same bus( I had a lot of problems with this) other people are on it to solve this as well. It sound like such a nice p&p board. Arduino is not the same as the teensy.

ziatajo, 2021-03-13:
Hi evermore99,

Thank you for your fast reply!
What do you mean with two seperate SPI's? I only have the Eval Board connected to the ESP. I don't think something else is using the SPI than? (or is USB using SPI?)

evermore99, 2021-03-13:
Also the ad7124 max speed is 5MHz I believe on clk speed. Tft Screens can go faster

BastiaanMDC, 2021-03-13:
The usb runs on the Rs232 bus. That’s different.
SPI = serial pheriphal interface stands for full duplex, meaning sending and receiving in both directions simultaneously. Look at the Pinot for the teensy3.2 4.0 or 4.1,3.6 3.5 all have also a Mosi miso CS DC and a CLK. Also they have a MiSO1 and a MOSI and SCK1 and some have a third bus.

Use one bus for a display and the other for the ad7124. Test that first. And see if you still have problems.

I haven’t yet worked with the esp32 so I can’t tell.

Outlook voor iOS<https://aka.ms/o0ukef> downloaden

evermore99, 2021-03-13:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html

evermore99, 2021-03-13:
Read the warning:)

ziatajo, 2021-03-13:
Thank four your answers. But I dont have a TFT screen or anything connected.
Just this board: [https://wiki.analog.com/resources/eval/user-guides/circuits-from-the-lab/eval-ad7124-8-pmdz](url)
to this one: [https://learn.adafruit.com/adafruit-huzzah32-esp32-feather/pinouts](url)

ziatajo, 2021-03-15:
Ok, I resoldered it to my Teensy 3.2.
However, it seems to be the same problem. I have around 160Hz with a Filterword of 120, but only 180Hz with a Filterword of 60.
But I did some more tests now. 
I did some timing measurements (I dont have an oscilloscope, have to use micros() ):
` a=micros();
   strainADC.waitEndOfConversion(500);
  b=micros();    
  
dataWord = strainADC.getData();  
ch = strainADC.currentChannel();`
 
waitEndOfConversion takes sometimes 2100us, most of the time 4200us. if I reduce the 500 to 1, it always takes 2100us
detData takes 1100us
currentChannel takes also 1100us

Do you have an idea what I could change?

msuciu, 2021-03-23:
Hi,
Please check out here https://github.com/analogdevicesinc/EVAL-ADICUP3029/tree/master/projects/ADuCM3029_demo_ad7124_8PMDZ a possible solution for your problem.

ziatajo, 2021-03-23:
Hi, thank you! Any idea where to start checking? there are a lot of documents :D

msuciu, 2021-03-24:
Hi,
You can start from here:
https://github.com/analogdevicesinc/no-OS/tree/master/projects/ad7124-8pmdz/src
https://github.com/analogdevicesinc/no-OS/blob/master/projects/ad7124-8pmdz/src.mk

You can also start using Cross Core Embedded Studio to see how the demo project looks like and to follow the software structure more easily or if you have an EVAL-ADICUP3029 around you can start to develop your application starting from the demo project and then port it to another platform.
https://wiki.analog.com/resources/no-os/build

Also, feel free to ask more questions on EnginnerZone https://ez.analog.com/ to reach out to more people who can help you.

RicRea, 2021-06-11:
Hi guys,

I'm facing a similar problem using the library provided here (https://github.com/epsilonrt/ad7124) for Arduino Uno and a MIKROE ADC-6 click board (embedding an AD7124-8).

From the datasheet of the ADC, in full power mode, the SPS with a FilterWord of 1 should be the maximum, 19200, without post filters and using 1 single channel. 
Thus I suppose that, for example, with FW=1 and using 12 channels, in continuous conversion mode,  I should achieve roughly 19200/12 = 1600 SPS for each channel, while I'm getting something lower that 50!

Thus, the system acts as if it were some bottleneck that reduce the acquisition rate dramatically. 

I had a look inside the library files but everything seems correct (at least for me, not an expert). 

Any kind of help, questions, tips that could bring some useful idea in mind to solve that issue is really appreciated.

Thanks!